forntoh/LcdMenu #2: Using LiquidCrystal without I2C

jcvillegasfernandez, 2020-07-30:
I could figure out how to make it work with LiquidCrystal.h without I2C, but I could not make work for both with a #define directive
example #define NO_I2C for LiquidCrystal.h

Could you adapt you version for both systems "LiquidCrystal.h" and "LiquidCrystal_I2C.h"

forntoh, 2020-07-30:
Thanks @jcvillegasfernandez, I will work on that next week, It wasn't my priority.
If you can adapt it earlier submit a PR. Contributions are accepted. 😊

jcvillegasfernandez, 2020-07-30:
I do not know how to submit a PR, I tried but I couln't.
But I enclose de files as text, very few changes.
file LcdNOI2Cmenu.h
/*
  LcdNOI2CMenu.h - Main include file for the LcdMenu Library

  MIT License

  Copyright (c) 2020 Forntoh Thomas

  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files (the "Software"), to deal
  in the Software without restriction, including without limitation the rights
  to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
  copies of the Software, and to permit persons to whom the Software is
  furnished to do so, subject to the following conditions:

  The above copyright notice and this permission notice shall be included in all
  copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
  IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
  AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
  OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
  SOFTWARE.
*/

#ifndef LcdMenu_H
#define LcdMenu_H

#include <Arduino.h>
#include <LiquidCrystal.h>   // LINE CHANGED

#include "MenuItem.h"

/**
 * the LcdMenu class
 */
class LcdMenu {
   private:
    /**
     * Cursor position
     */
    uint8_t cursorPosition = 1;
    /**
     * First visible item's position in the menu array
     */
    uint8_t top = 1;
    /**
     * Last visible item's position in the menu array
     */
    uint8_t bottom;
    /**
     * Rows on the LCD Display
     */
    uint8_t maxRows;
    /**
     * Columns on the LCD Display
     */
    uint8_t maxCols;
    /**
     * Array of menu items
     */
    MenuItem* currentMenuTable;
    /**
     * Down arrow (â†“)
     */
    byte downArrow[8] = {
[... 586 lines cut ...]
    if (millis() == startTime + duration) paint();
}
/**
 * Get a `MenuItem` at position
 *
 * @return `MenuItem` - item at `position`
 */
MenuItem* LcdMenu::getItemAt(uint8_t position) {
    return &currentMenuTable[position];
}
/**
 * Places the cursor at end of Menu's text.
 *
 * @param item MenuItem where the cursor should be placed
 * @relatesalso MenuItem
 */
void LcdMenu::placeCursorAtEnd(MenuItem *item) {
    uint8_t col = ((String)item->getText()).length() + 2 + item->value.length();
    lcd->setCursor(constrain(col, 0, maxCols - 1), cursorPosition - top);
}

And finally in your example change this line 
  menu.setupLcdWithMenu(0x27, mainMenu); 
for this one
  menu.setupLcdWithMenu( 8, 9, 4, 5, 6, 7, mainMenu);

jcvillegasfernandez, 2020-07-31:
Finally, I was able to find out how to do it using a #define NO_I2C_LCD, but I still don't know how to open a PR, if you want I can send the files to an email

forntoh, 2020-07-31:
- You can need to fork the repository (at the top right corner of the screen), It creates your own copy on your GitHub account
- Clone it to your PC
- Modify it
- Push the changes to the forked repo
- When you open the repo on your browser, GitHub will propose that you should make a Pull Request
- That's all

jcvillegasfernandez, 2020-08-02:
Thanks for your help I have opened a PR

forntoh, 2020-08-07:
> Thanks for your help I have opened a PR

Please check this [review](https://github.com/forntoh/LcdMenu/pull/3#discussion_r466424047)

closed 2020-11-14 by forntoh