forntoh/LcdMenu #5: ItemInput callback issue

kulverstukas1, 2020-10-19:
In the examples it's written that callbacks are available for `ItemCommand` and `ItemToggle`, however in the code I see that `ItemInput` can also have a callback and even calls it with `.enter()`, but every time I do that, Arduino resets completely without firing the callback.
I'm developing on an Arduino Nano.

This is my menu code:

<pre>
MenuItem timerMenu[] = {
  ItemSubHeader(mainMenu),
  MenuItem((char*)"Delay time"),
  ItemInput((char*)"", (char*)TIME_ZERO, timerCallback),
  MenuItem("Game time"),
  ItemInput("", TIME_ZERO, NULL),
  ItemFooter()
};
</pre>

timerCallback() is just this:

<pre>
void timerCallback() {
  Serial.println("DONE");
}
</pre>

Same if I have `NULL` as a callback as well.

forntoh, 2020-10-19:
👍 Thanks for reporting this issue, I will check it ASAP

kulverstukas1, 2020-11-12:
Any updates on this?
Btw I'm also getting these compile errors all the time:

```
In file included from .pio\libdeps\NanoaATmega328\LcdMenu\src/LcdMenu.h:33:0,
                 from src\main.cpp:6:
.pio\libdeps\NanoaATmega328\LcdMenu\src/MenuItem.h:47:18: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     char* text = "";
                  ^
.pio\libdeps\NanoaATmega328\LcdMenu\src/MenuItem.h:84:20: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     char* textOn = "ON";
                    ^
.pio\libdeps\NanoaATmega328\LcdMenu\src/MenuItem.h:85:21: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     char* textOff = "OFF";
                     ^
.pio\libdeps\NanoaATmega328\LcdMenu\src/MenuItem.h: In constructor 'MenuItem::MenuItem(char*, char*, fptr, byte)':
.pio\libdeps\NanoaATmega328\LcdMenu\src/MenuItem.h:86:20: warning: 'MenuItem::value' will be initialized after [-Wreorder]
     String value = "";
                    ^
In file included from c:\users\kulverstukas\.platformio\packages\toolchain-atmelavr\avr\include\avr\eeprom.h:50:0,
                 from C:\users\kulverstukas\.platformio\packages\framework-arduino-avr\cores\arduino/USBAPI.h:25,
                 from C:\users\kulverstukas\.platformio\packages\framework-arduino-avr\cores\arduino/Arduino.h:233,
                 from src\main.cpp:1:
.pio\libdeps\NanoaATmega328\LcdMenu\src/MenuItem.h:48:21: warning:   'void (* MenuItem::callback)()' [-Wreorder]
     fptr callback = NULL;
                     ^
In file included from .pio\libdeps\NanoaATmega328\LcdMenu\src/LcdMenu.h:33:0,
                 from src\main.cpp:6:
.pio\libdeps\NanoaATmega328\LcdMenu\src/MenuItem.h:65:5: warning:   when initialized here [-Wreorder]
     MenuItem(char* text, char* value, fptr callbac, byte type)
     ^
.pio\libdeps\NanoaATmega328\LcdMenu\src/MenuItem.h:65:5: warning: 'MenuItem::callback' is initialized with itself [-Winit-self]
```

forntoh, 2020-11-12:
Which Arduino hardware are you using? @kulverstukas1

kulverstukas1, 2020-11-12:
Arduino Nano (ATmega328p)

forntoh, 2020-11-12:
I'll try to compile it again on the Arduino Uno and see if I get those warnings.

forntoh, 2020-11-12:
@kulverstukas1 Please download the **master** branch and test if the above issues have been solved, after that, I can create a release that can be downloaded from Arduino Libraries

kulverstukas1, 2020-11-12:
Yes, that did it! callbacks are now fired. If I set the callback as NULL, Arduino still resets though.
Initially looking at the source I didn't think it could be a typo lol
The compile warnings are still there though.

forntoh, 2020-11-12:
Those warning messages actually showed me that it was a typo, thanks a lot
I don't see those warnings when I compile 🤷‍♀️

forntoh, 2020-11-12:
I couldn't Identify the callback issue 25 days ago, and finally 😊

kulverstukas1, 2020-11-12:
Saved by the stacktrace!
Interesting about the errors. I'm also using Platformio IDE, might be that it shows more verbose output than the Arduino IDE if that's what you're using.

forntoh, 2020-11-12:
I use `Visual Studio Code` which actually uses the Arduino instance installed on the PC 🤷‍♀️

kulverstukas1, 2020-11-12:
Well, I use Visual Studio Code too, Platformio is like an add-on for that which is very awesome. Maybe you could try it and see [https://platformio.org/](https://platformio.org/) but anyway, those are warning and out-of-scope with this issue. I can open a separate issue for it.

forntoh, 2020-11-12:
@kulverstukas1 Arduino will no longer restart when the callback is `NULL`. I added a `NULL `checker where the callback is called

kulverstukas1, 2020-11-12:
Tested. Restarts no more.

closed 2020-11-13 by forntoh

forntoh, 2020-11-14:
You can download the latest version on the **Arduino Library Manager** : [what's new?](https://github.com/forntoh/LcdMenu/releases/tag/1.1.0)