forntoh/LcdMenu #8: Go back with rotary encoder

dj-fiorex, 2020-12-03:
Hi all, i'm trying your library with a rotary encoder with a button (KY-040), i can go up, down and enter sub-menu, nut i can't go back to the main menu because ItemSubHeader is not visible in the submenu.
Can you help me?
MYCODE: 
```
#include <Arduino.h>

#include <LcdMenu.h>

#define LCD_ROWS 4
#define LCD_COLS 16

//Definizione pin
#define PIN_CLK 9
#define PIN_DT 10
#define PIN_SW 11

// Conserva stato ultima posizione encoder
volatile int prevClk;
volatile int prevDt;

volatile int pressingButton = 0;

extern MenuItem mainMenu[];
extern MenuItem settingsMenu[];

MenuItem mainMenu[] = {ItemHeader(),
                       MenuItem("Start service"),
                       MenuItem("Connect to WiFi"),
                       ItemSubMenu("Settings", settingsMenu),
                       MenuItem("Blink SOS"),
                       MenuItem("Blink random"),
                       ItemFooter()};

/**
 * Create submenu and precise its parent
 */
MenuItem settingsMenu[] = {ItemSubHeader(mainMenu),
                           MenuItem("Backlight"),
                           MenuItem("Contrast"),                        
                           ItemFooter()};

LcdMenu menu(LCD_ROWS, LCD_COLS);

void doEncoderA()
{

  Serial.println("doEncoderA");

  // Leggi valori dai pin dell'encoder
  int currClk = digitalRead(PIN_CLK);
  int currDt = digitalRead(PIN_DT);

  // Se la lettura corrente è diversa dall'ultima memorizzata
  if (currClk != prevClk)
  {

    Serial.print("[prevClk, prevDt | currClk, currDt] = ");
    Serial.print(prevClk);
    Serial.print(" ");
    Serial.print(prevDt);
    Serial.print(" | ");
    Serial.print(currClk);
    Serial.print(" ");
    Serial.print(currDt);

    //Se il valore di dt è uguale al valore corrente di clk
    if (currDt == currClk)
    {
      Serial.print(" -");
      menu.up();
    }
    else
    {
      Serial.print(" +");
      menu.down();
    }

    // Mostra lo stato del contatore
    Serial.print(" = ");

    //Aggiorna valori
    prevClk = currClk;
    prevDt = currDt;
  }
}
void doEncoderB()
{
  Serial.println("doEncoderB");
}
void encoderButtonPressed()
{
  if (pressingButton > 0)
  {
    pressingButton = 0;
  }
  else
  {
    Serial.println("ButtonPressed");
    menu.enter();
    pressingButton = 1;
  }
}

void setupRotaryEncoder()
{
  pinMode(PIN_CLK, INPUT);
  pinMode(PIN_DT, INPUT);
  pinMode(PIN_SW, INPUT);

  // Encoder pins on interrupt
  attachInterrupt(digitalPinToInterrupt(PIN_CLK), doEncoderA, CHANGE);
  attachInterrupt(digitalPinToInterrupt(PIN_DT), doEncoderB, CHANGE);
  attachInterrupt(digitalPinToInterrupt(PIN_SW), encoderButtonPressed, RISING);

  // Esegui prima lettura dei valori sui pin
  prevClk = digitalRead(PIN_CLK);
  prevDt = digitalRead(PIN_DT);
}

void setup()
{

  Serial.begin(9600);
  setupRotaryEncoder();
  menu.setupLcdWithMenu(0x27, mainMenu);
}

void loop()
{

}

```
Thanks

dj-fiorex, 2020-12-03:
Ok my fault! I just added to the submenu declaration an ItemSubMenu element that point to the main menu!
Thanks anyway!

closed 2020-12-03 by dj-fiorex

forntoh, 2020-12-04:
@dj-fiorex Glad you found the problem, don't hesitate to report any issues you have. I will be including more features in the library soon