forntoh/LcdMenu #15: [NEW] Add LCD timeout

forntoh, 2021-03-16:
The library should include an optional timeout.

```cpp
void setTimeout(uint8_t timeout);
```

- When the time elapses, the LCD is cleared and the backlight is switched off.
- The LCD's state will be restored when a key is pressed _(or if action invokes `paint()`)_.
- When the timeout is set to zero, the LCD stays on as long as it is powered.
- If no timeout is provided, a default of **5s** is used.

### Benefit
- Save energy 🤷‍♂️🚮⚡

strikar21, 2021-12-13:
currentMillis = millis();

  if (currentMillis - startMillis >= period) menu.lcd->setBacklight(LOW);
  if (button != '0') {
    menu.lcd->setBacklight(HIGH);
    startMillis = millis();
  }

inside the void loop function
unsigned long currentMillis;
unsigned long startMillis;
unsigned long period = 10*1000; //( FOR 10 second Delay)

however, i cannot get the text to be displayed after backlight comes ON.
in the sense, backlight goes away, 
when i press a button, i'd like the main menu to come back on, instead of the text already there on the screen.
anyway to do it?

forntoh, 2023-04-04:
@jiriks74 Any progress with this? 🙂

PR #104 (2023-06-16): Added LCD Timeout
closes #15
closed 2023-06-16 by forntoh