forntoh/LcdMenu #21: Backlight support for non-I2C

thijstriemstra, 2021-06-07:
I had to hack and add the last 2 parameters of `new LiquidCrystal` in order to get the backlight working.

```
#ifndef USE_STANDARD_LCD
        uint8_t lcd_Addr, MenuItem* menu) {
        lcd = new LiquidCrystal_I2C(lcd_Addr, maxCols, maxRows);
        lcd->init();
        lcd->backlight();
#else
        uint8_t rs, uint8_t en, uint8_t d0, uint8_t d1, uint8_t d2, uint8_t d3,
        MenuItem* menu) {
        this->lcd = new LiquidCrystal(rs, en, d0, d1, d2, d3, DisplayBacklightPin, POSITIVE);
        this->lcd->begin(maxCols, maxRows);
#endif
```

would be nice if this could be supported in the library.

forntoh, 2021-06-11:
What are the possible values for `DisplayBacklightPin`, `POSITIVE`?
Are you using the official `LiquidCrystal` library by Arduino?

thijstriemstra, 2021-06-13:
I'm using `fmalpartida/LiquidCrystal` and `DisplayBacklightPin` is a var from my project with value 5. `POSITIVE` is a constant from that LiquidCrystal library. e.g. `t_backlightPol pol = POSITIVE`.

thijstriemstra, 2023-03-31:
Should I make PR for this fix @forntoh?

thijstriemstra, 2023-03-31:
> Should I make PR for this fix @forntoh?

see #75

closed 2023-08-12 by github-actions[bot]