forntoh/LcdMenu #51: Enhancement: ability to print custom characters

jiriks74, 2022-01-26:
Hi,
I'd like the ability to print custom characters (my language has letters like č,ř,ě,š, etc.). My idea I have ATM is that you'd have a field where you'd assign the characters to a number to a created character. A bodge but it would work. 

I'll try to help, but I'm not a good coder yet.

forntoh, 2022-01-31:
I believe that this doesn't depend on the library, it depends on the LCD module you use. Each one has its supported charset.
e.g. on page 3 of [this document](https://www.lcd-module.com/fileadmin/eng/pdf/doma/dip204-4e.pdf) you can see the supported charset for this panel.

Did you try to put the values in string e.g. `ItemInput("čřěš", callback)` and check if it works?

jiriks74, 2022-01-31:
> I believe that this doesn't depend on the library, it depends on the LCD module you use. Each one has its supported charset. e.g. on page 3 of [this [...]
The problem is with the compiler from my understanding. It works in ASCII but those charecters are unicode.

I've got around it in past by creating a custom character in the LCD library (the same that is used in this one) and I've just made a function that parsed the text (it checked for 2 characters right after each other - the unicode characters were translated into 2 ascii, so I figured them out and checked for them). If the function met those characters, it would print the custom character, if not it would print whatever was in the string.

closed 2023-06-15 by forntoh