forntoh/LcdMenu #64: not taking EEPROM data

tm-sanjay, 2022-06-28:
When the menu item is created, the initial data is assigned to the variable. But If I try to assign an EEPROM data it's not updating, it is taking the initial data only.
If I change the value from the Menu it is updating, but if I manually update the data outside the Menu it is not taking.

This will work
For ex:
```
//if this is my menu Item
ItemInput("ValueA", String(valueA), inputCallback)

//Inside inputCallback
if (selectedMenu == "ValueA") {
    valueA = value.toFloat();
    Serial.println("valueA: " + String(valueA));
  }
````

If I try to update "valueA" outside the **inputCallBack** it is not working
Something like this:
```
void updateA() {
  valueA = 10;
  menu.show();
}
```
Same for EEPROM data.

forntoh, 2022-06-29:
Please look the answer to #9

tm-sanjay, 2022-06-29:
Thank you for the reference.
I was able to change using ```menu.getItemAt(0)->value  = "New value"```
But not with this one ```menu.[``your position``][``your sub position``]->value = "your new value here"```
I was getting the **expected a member name** error, this is not allowed.

forntoh, 2022-07-05:
```menu.getItemAt(0)->value  = "New value"``` is equivalent to  ```menu[0]->value  = "New value"```
Did you try this? ```menu[``your position``][``your sub position``]->value = "your new value here"``` only works if the menu item which you want to access is in a submenu (not in the main menu).
The first index `[]` is for the submenu's position in the main menu and the second index `[]` is for the item index in the submenu.

If indeed your item is in a sub-menu, I will have a look at the implementation to see if I made some errors

tm-sanjay, 2022-07-05:
```menu[0]->value = "New value"``` works
but not ```menu[``your position``][``your sub position``]->value = "your new value here"```

This is the error I'm getting
[image]

forntoh, 2022-07-06:
Try this `menu[1][2].value`, I think that it should work this way, because, from the code, the second operator doesn't return a pointer.

tm-sanjay, 2022-07-06:
Thank you so much, this is working. I spent one week achieving this output. But I created a new library for the menu in that process : )