forntoh/LcdMenu #66: Newliquidcrystal_1.3.5 and LcdMenu - compatibility issue

Stefano-69, 2022-10-18:
Hi Thomas
Your library seem exactly was I need, but I have a problem.
I changed to use the I2C library into the LiquidMenu_config.h but I have an error at the line " lcd.init();)" ->int LiquidCrystal_I2C::init()' is private within this context". I expect that the library "Newliquidcrystal_1.3.5" is not compatible with your code.
I tryed to cancel it, but nothing works. 
May you tell me which library is to use with your code? Thank you in advance and best regards,
Stefano

Stefano-69, 2022-10-18:
I found a replacement library called "liquidcrystal_I2C" that works!

forntoh, 2022-10-20:
You can look at the library dependency for the exact library I used
[platformio.ini](https://github.com/forntoh/LcdMenu/blob/8bf6dee0830df5f2d973ab933813f54e2bf1524b/platformio.ini)

closed 2022-10-20 by forntoh