forntoh/LcdMenu #90: [BUG] -

Cserni, 2023-06-03:
**Describe the bug**
Hello
I really like what LcdMenu has to offer. But unfortunately, the example codes return with a library error when compiled. Menu_basic errors here
Arduino: 1.8.13 (Windows 7), Motherboard: "Arduino Pro or Pro Mini, ATmega328P (5V, 16 MHz)"

In file included from C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/LcdMenu.h:34:0,

                  from C:\Users\bandi\Documents\Arduino\menu_basic\menu_basic.ino:1:

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h: In member function 'virtual void MenuItem::setIsOn(boolean)':

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h:116:34: warning: unused parameter 'isOn' [-Wunused-parameter]

      virtual void setIsOn(boolean isOn){};

                                   ^~~~

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h: In member function 'virtual void MenuItem::setValue(String)':

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h:120:34: warning: unused parameter 'value' [-Wunused-parameter]

      virtual void setValue(String value){};

                                   ^~~~~

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h: In member function 'void MenuItem::setText(const char*)':

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h:125:30: warning: unused parameter 'text' [-Wunused-parameter]

      void setText(const char* text){};

                               ^~~~

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h: In member function 'virtual void MenuItem::setCallBack(fptr)':

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h:130:35: warning: unused parameter 'callback' [-Wunused-parameter]

      virtual void setCallBack(fptr callback){};

                                    ^~~~~~~~

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h: In member function 'virtual void MenuItem::setItemIndex(uint8_t)':

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h:134:39: warning: unused parameter 'itemIndex' [-Wunused-parameter]

      virtual void setItemIndex(uint8_t itemIndex){};

                                        ^~~~~~~~~

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h: In constructor 'ItemHeader::ItemHeader(const char*, MenuItem**, byte)':

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h:169:28: warning: unused parameter 'text' [-Wunused-parameter]

      ItemHeader(const char* text, MenuItem** parent, byte type)

                             ^~~~

C:\Users\bandi\Documents\Arduino\menu_basic\menu_basic.ino: At global scope:

C:\Users\bandi\Documents\Arduino\libraries\LcdMenu-3.0.0\src/MenuItem.h:212:58: error: expected primary-expression before ',' token

      MenuItem* mainMenu[] = {new ItemHeader(), __VA_ARGS__, new ItemFooter()}

                                                           ^

C:\Users\bandi\Documents\Arduino\menu_basic\menu_basic.ino:31:1: note: in expansion of macro 'MAIN_MENU'

  MAIN_MENU(

  ^~~~~~~~~

exit status 1

Error translating to Arduino Pro or Pro Mini motherboard.

This report could have more information
"Show more output when compiling"
by enabling the option in File -> Preferences.

What do I do wrong?

forntoh, 2023-06-03:
Try using the latest version https://www.arduino.cc/en/software
If a previous version of the lib doesn't work either, check this https://forum.arduino.cc/t/error-compiling-for-board-arduino-pro-or-pro-mini-solved-see-lessons-learned/645202/10

Cserni, 2023-06-03:
Thanks
Arduino IDE 1.8.19 update helped :-)

closed 2023-06-04 by forntoh