forntoh/LcdMenu #96: SubMenu problem

Cserni, 2023-06-13:
### What happened?

Dear Thomas
I am asking for your help again. While testing SubMenu.ino, I noticed that the submenu text does not appear on the lcd when running SubMenu. This is also the case if I call the submenu from a submenu. I don't know if the problem is known. I couldn't find it on the github pages. The menu item syntax is good. The program editor does not give any errors, there are no translation errors. I looked through the .h files, but I don't know what I'm looking for. I thought it might be a type conflict, but I couldn't find any indication of this, or a version update...? Please deal with the problem.

#include <ItemSubMenu.h>
#include <LcdMenu.h>
Arduino: 1.8.19 (Windows 7), Motherboard: "Arduino Pro or Pro Mini, ATmega328P (5V, 16 MHz)", LcdMenu version 3.0.0
Postscript:
I promise I'll close the topic when it's finished :)
Hi
Andrew

### Version

3.0.0

### What environment are you seeing the problem on?

Arduino

### Board type

Arduino Pro or Pro Mini, ATmega328P (5V, 16 MHz)

forntoh, 2023-06-13:
I'll have a look and try to reproduce the issue.

maansch, 2023-06-14:
Hello,
I have the same problem, the submenu name is not displayed.
gr
Marc-Andre

Code: 
example SubMenu.ino

environment :
Arduino: 1.8.19 (Linux), 

Board Typ:
AZ-Delivery ESP-32Development-Boards
ESP-WROOM-32 Prozessor  160MHz Tensilica L108 32 bit Dual-Core CPU
» 512 KB SRAM & 16 MB Flashspeicher

LcdMenu Version 3.0.0

closed 2023-06-16 by forntoh

maansch, 2023-06-18:
tested with v3.1.1 , nothing changed

forntoh, 2023-06-18:
Release is still planned

maansch, 2023-06-18:
tested with v3.2.0 , it works :+1: 

thanks

Cserni, 2023-06-22:
Thanks, it works for me too 😀✌️💯

maansch ***@***.***> ezt írta (időpont: 2023. jún. 18., V
12:47):

PR #102 (2023-06-16): Fixed ItemSubMenu text not displayed
fixes #96