forntoh/LcdMenu #98: Issue with Charsetinput.ino on esp32

maansch, 2023-06-14:
### What happened?

Hi Thomas,
try to use the LcdMenu for my project I run in a couple of issues.

Test with the Charsetinput. Ino example, that’s happened:

(using Core Version 1.6.0 )
After select the first digit and hit enter “nan” shows up on lcd row.

(using Core Version 2.9.0 )
After select the first digit and hit enter “task_wdt” shows up on lcd row.

(both Versions)

BACKSPACE can’t delete the “nan”, just override is possible.

Select first digit again -> enter works -> select second digit -> enter works => leaving the Input mode => result is 33n.
the "n" is from "nan" -> third digit was not edited.

br
Marc-Andre

Code: example Charsetinput.ino

environment : Arduino: 1.8.19 (Linux),

Board Typ: esp32 /AZ-Delivery ESP-32Development-Board, ESP-WROOM-32 Prozessor 160MHz, Tensilica L108 32 bit Dual-Core CPU,  512 KB SRAM & 16 MB Flashspeicher.

Bord driver for testing:
Core Version 2.9.0 
Core Version 1.6.0 

LcdMenu Version 3.0.0 / 3.0.1

### Version

3.0.0

### What environment are you seeing the problem on?

Arduino

### Board type

ESP-32 Development-Board

### Relevant log output

```shell
none
```

forntoh, 2023-06-14:
I am aware of this issue and will fix it soon.

closed 2023-06-15 by forntoh

maansch, 2023-06-16:
worked with 3.1.0
Thanks