forntoh/LcdMenu #100: Error:  compiling  example List.ino esp32

maansch, 2023-06-15:
### Discussed in https://github.com/forntoh/LcdMenu/discussions/99

<div type='discussions-op-text'>

<sup>Originally posted by **maansch** June 14, 2023</sup>
Hi Thomas,
try to use the List.ino, during compiling I receive these error msg.

/home/mas/A_Prog_A/Arduino/libraries/LcdMenu/src/ItemList.h: In constructor 'ItemList::ItemList(const char*, String*, uint8_t, fptrInt)':
/home/mas/A_Prog_A/Arduino/libraries/LcdMenu/src/ItemList.h:34:21: error: 'ItemList::items' will be initialized after [-Werror=reorder]
     String* items = NULL;     ///< Pointer to an array of items
                     ^
/home/mas/A_Prog_A/Arduino/libraries/LcdMenu/src/ItemList.h:33:24: error:   'void (* ItemList::callback)(uint8_t)' [-Werror=reorder]
     fptrInt callback = NULL;  ///< Pointer to a callback function
                        ^
In file included from /tmp/arduino_modified_sketch_881776/List.ino:21:0:
/home/mas/A_Prog_A/Arduino/libraries/LcdMenu/src/ItemList.h:47:5: error:   when initialized here [-Werror=reorder]
     ItemList(const char* key, String* items, const uint8_t itemCount,
     ^
cc1plus: some warnings being treated as errors

Bibliothek LcdMenu in Version 3.0.1 im Ordner: /home/mas/A_Prog_A/Arduino/libraries/LcdMenu  wird verwendet
Bibliothek LiquidCrystal_I2C in Version 1.1.4 im Ordner: /home/mas/A_Prog_A/Arduino/libraries/LiquidCrystal_I2C  wird verwendet
Bibliothek Wire in Version 1.0.1 im Ordner: /home/mas/arduinoide/arduino-1.8.19_ESP32/portable/packages/esp32/hardware/esp32/1.0.6/libraries/Wire  wird verwendet

exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.

Do you have an idea where the error could be?
I would be grateful for your help.

br
Marc-Andre

Code: example List.ino
environment : Arduino: 1.8.19 (Linux),
Board Typ: esp32
AZ-Delivery ESP-32Development-Board, ESP-WROOM-32 Prozessor 160MHz, Tensilica L108 32 bit Dual-Core CPU,  512 KB SRAM & 16 MB Flashspeicher.
Bord driver for testing:
Core Version 2.9.0 
Core Version 1.6.0 </div>

maansch, 2023-07-02:
I still have hints when compiling.

...Arduino/libraries/LcdMenu/src/ItemList.h: In member function 'virtual void ItemList::setItemIndex(uint8_t)':
.../.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Arduino.h:71:39: warning: comparison is always false due to limited range of data type [-Wtype-limits]
#define constrain(amt,low,high) ((amt)<(low)?(low):((amt)>(high)?(high):(amt)))
______________________________^

.../Arduino/libraries/LcdMenu/src/ItemList.h:67:27: note: in expansion of macro 'constrain'
         this->itemIndex = constrain(itemIndex, 0, itemCount - 1);
____________ ^

forntoh, 2023-07-31:
Unable to reproduce this with the latest version of the library

PR #101 (2023-06-16): Fixed compile warnings related to ItemList
fixes #100
closed 2023-07-31 by forntoh