forntoh/LcdMenu #128: ITEM_COMMAND not working

rozrabiak, 2023-08-27:
### What happened?

```cpp
ITEM_COMMAND("EXIT MENU", exitMenu));  //close menu and off backlight

void exitMenu()
{
  Serial.println("working");
  menu.resetMenu();
  menu.hide();
}
```

And function not invoking when presing ENTER.
But I can navigate throught menu, I can enter to submenu using ENTER.

Arduino & STM32F4.
Any ideas?

### Version

3.3.2

### What environment are you seeing the problem in?

Platform IO

### Board type

blackpill STM32F4ccu6

forntoh, 2023-08-28:
Please make sure that your imports are correctly placed

see https://lcdmenu.forntoh.dev/usage

closed 2023-08-28 by forntoh

hapiel, 2023-10-07:
This could be marked more clearly, I've been debugging for 3 hours before I found this!