forntoh/LcdMenu #133: [BUG]: Can't compile if I add #include "LcdMenu.h" to any other file than main.cpp

hapiel, 2023-09-26:
### What happened?

in my platformio project, if I add #include <LcdMenu.h>to another .h file, I get a conflict error when compiling.

It appears to only happen if the .h is included in a file that contains a constructor with at least 1 argument. Very strange.

Error:

`c:/users/daniel/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\main\lib4fd\libmenu.a(MenuClass.cpp.o): in function `substring(char*, unsigned char, unsigned char, char*)':
D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:4: multiple definition of `substring(char*, unsigned char, unsigned char, char*)'; .pio\build\main\src\main.cpp.o:D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:4: first defined here
c:/users/daniel/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\main\lib4fd\libmenu.a(MenuClass.cpp.o): in function `concat(char const*, char, char const*, char*)':
D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:9: multiple definition of `concat(char const*, char, char const*, char*)'; .pio\build\main\src\main.cpp.o:D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:9: first defined here
c:/users/daniel/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\main\lib4fd\libmenu.a(MenuClass.cpp.o): in function `concat(char const*, char, char*)':
D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:19: multiple definition of `concat(char const*, char, char*)'; .pio\build\main\src\main.cpp.o:D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:19: first defined here
c:/users/daniel/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\main\lib4fd\libmenu.a(MenuClass.cpp.o): in function `concat(char const*, char const*, char*)':
D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:26: multiple definition of `concat(char const*, char const*, char*)'; .pio\build\main\src\main.cpp.o:D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:26: first defined here
c:/users/daniel/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\main\lib4fd\libmenu.a(MenuClass.cpp.o): in function `remove(char*, unsigned char, unsigned char)':
D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:31: multiple definition of `remove(char*, unsigned char, unsigned char)'; .pio\build\main\src\main.cpp.o:D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:31: first defined here
c:/users/daniel/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\main\lib4fd\libmenu.a(MenuClass.cpp.o): in function `mapProgress(unsigned short, long, long)':
D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:39: multiple definition of `mapProgress(unsigned short, long, long)'; .pio\build\main\src\main.cpp.o:D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:39: first defined here
c:/users/daniel/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\main\lib4fd\libmenu.a(MenuClass.cpp.o): in function `mapProgress(unsigned short, float, float)':
D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:44: multiple definition of `mapProgress(unsigned short, float, float)'; .pio\build\main\src\main.cpp.o:D:\9mediatech\acrobotics\code\jona\Acrobot/.pio/libdeps/main/LcdMenu/src/utils.h:44: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\main\firmware.elf] Error 1`

To recreate:
main.cpp:
```cpp
#include "MenuClass.h"

MenuClass menuClass(0x27);

void setup(){
}

void loop(){
}
```

MenuClass.h:
```cpp

#ifndef MENUCLASS_H
#define MENUCLASS_H

#include <LcdMenu.h>

class MenuClass {
  public:
    MenuClass(uint8_t address);
  private:
    uint8_t address;
};

#endif
```

MenuClass.cpp:
```cpp
#include "MenuClass.h"

MenuClass::MenuClass(uint8_t address) {
  this->address = address;
}
```

### Version

3.3.2

nikensss, 2023-09-26:
first note, let's not call classes `Class`. Like `MenuClass`, it should just be `Menu`. Because otherwise, in order to be consistent, you should rename `address` to `addressUint_8t`.

Regarding the error, it says "multiple definition of `substring". There's probably to `substring` functions in scope and the compiler can't tell which one to use, have you tried fully qualifying the path of the function?

hapiel, 2023-09-26:
Even if I call no functions at all, I only include the LcdMenu.h, I already get the error.. Unless I include it in the main

forntoh, 2023-10-09:
Will investigate further, but it seems to work when you don't use a cpp file (ie define everything in the .h file)

```cpp

#ifndef MENUCLASS_H
#define MENUCLASS_H

#include <LcdMenu.h>

class MenuClass {
   private:
    uint8_t address;

   public:
    MenuClass(uint8_t address) : address(address) {}
};

#endif
```

closed 2023-10-09 by forntoh

hapiel, 2023-10-09:
I now made a temporary fix by making all the functions in utils `inline`.
See my fork:

github.com/hapiel/lcdMenu

forntoh, 2023-10-09:
This has been fixed in v3.3.3

PR #135 (2023-10-09): Separated utils definition from declaration
Fixes #133