forntoh/LcdMenu #168: [BUG]: Error installing LcdMenu

ZanPekosak, 2024-04-10:
### What happened?

When trying to install the library, I get the following issue. Installing the library with the ZIP method seems to work fine, but I can use Arduino simulators as the error shows up both in VS Code and in Arduino IDE, as well as online simulators, such as wokwi.

### Version

3.5.4

### What environment are you seeing the problem in?

Arduino

### Board type

Arduino Uno

### Relevant log output

```shell
Failed to install library: 'LcdMenu:3.5.4'. No valid dependencies solution found: dependency 'LiquidCrystal_I2C' is not available
```

ZanPekosak, 2024-04-11:
I have had the I2C library for a few years now. Tried to reinstall it and the issue still remains. I'll be home in about a week where I'll get access to my Arduino's, so I can do some further testing. But unzipping seems to work alright, which is what matters the most. Thank you help!

ArcticBeat05, 2024-06-07:
Getting same error on Arduino IDE 2.3.2 as well as on wokwi, the LiquidCrystal_I2C runs fine when LcdMenu is not installed, but once installed even the basic I2C example doesn't work and get error "Error installing LcdMenu: No valid dependencies solution found: dependency 'LiquidCrystal_I2C' is not available"

bigmike0640, 2024-06-25:
Same issue. I found the LiquidCrystal_I2C library and installed it and also manually installed this library. I get the following error when I try to compile just the LCDmenu Basic example.
.../LiquidCrystal_I2C/LiquidCrystal_I2C.h:182:7: note: declared private here

bigmike0640, 2024-06-26:
Update: The above experiment was running Arduino IDE on my macbook. I just did the same thing on my windows PC and it's working!

Filipsi, 2024-08-03:
I am also having this issue. Had to install LiquidCrystal_I2C library manually to fix this.

ZanPekosak, 2024-08-31:
UPDATE: Manual library installation works fine and does not cause issues with the I2C library on Arduino IDE. I haven't tried Wokwi yet, but apparently, the Arduino dev's have fixed the dependency issue. The library manager method of installation still doesn't work and will probably take some more time.

closed 2024-08-31 by ZanPekosak