forntoh/LcdMenu #203: [BUG]: Dependencies in the file "library.properties"

RoneFRANCE, 2024-09-17:
### What happened?

Hello,

Under wokwi using LCDMENU.h library which refers to LiquidCrystal_I2C library, I am getting this answer :
Error installing LcdMenu: No valid dependencies solution found: dependency 'LiquidCrystal_I2C' is not available

How could we get solved this problem on wokwi please ?

Thank you

Erwan

### Version

4.1.0

forntoh, 2024-09-17:
@RoneFRANCE Have you tried manually installing the LiquidCrystal_I2C library?

forntoh, 2024-09-17:
Please also check this conversation #168, the creator closed it so you might find a solution there.

RoneFRANCE, 2024-09-17:
I am not able to add libraries manually in wokwi as I don't have the paid version

closed 2024-09-19 by forntoh

forntoh, 2024-09-19:
@RoneFRANCE @ZanPekosak 

This should be solved in [v4.2.0](https://github.com/forntoh/LcdMenu/releases/tag/4.2.0) by #215, you will just need to install all dependencies manually.

PR #215 (2024-09-19): Remove dependencies [fix for #203]
Remove unnecessary dependencies from library properties.

fixes #203
commit 84be4f20 (2024-09-19): Remove dependencies [fix for #203] (#215)

Remove dependencies

Remove unnecessary dependencies from library properties.

fixes #203