forntoh/LcdMenu #217: [BUG]: DEBUG mode not working

forntoh, 2024-09-20:
### What happened?

Debug mode stopped working

### Version

4.2.0

### What environment are you seeing the problem in?

Platform IO

### Board type

AVR

ShishkinDmitriy, 2024-09-20:
Can `#define DEBUG` be added on top of `utils/utils.h`?
Right now the problem is that some classes have references to each other. And it's not possible to include each other.
That's why I have to move some implementations into `cpp`.

ShishkinDmitriy, 2024-09-20:
Think I found.
Need to add `build_flags = -D DEBUG` into `platformio.ini`.
It will be applied for all at the beginning.

forntoh, 2024-09-20:
That works, I'll add it and add logs to MenuScreen.

ShishkinDmitriy, 2024-09-20:
Probably it's possible to create multiple envs, one with, one without debug.

PR #220 (2024-09-20): [chore] Improved Debug Logging
Fixes #217

- Replaced `printCmd` with `printLog` for better debug messages
- Added debug flag in `platformio.ini` for debugging purposes

### Sample

```bash
#LOG# MenuScreen:down=1
#LOG# MenuScreen:down=2
#LOG# MenuScreen:down=3
#LOG# MenuScreen:down=3
#LOG# MenuScreen:up=2
#LOG# MenuScreen:up=1
#LOG# MenuScreen:up=0
#LOG# ItemSubMenu::enter=Set user
#LOG# MenuScreen:down=1
#LOG# MenuScreen:up=0
#LOG# ItemInput::enterEditMode=
#LOG# ItemInputCharset::down=A
#LOG# ItemInputCharset::down=B
#LOG# ItemInputCharset::down=C
#LOG# ItemInputCharset::up=B
#LOG# ItemInpu [... cut]
closed 2024-09-20 by forntoh