forntoh/LcdMenu #245: [NEW-FEATURE] - Support for esp32

almofada1, 2024-10-17:
### Describe the solution you'd like?

Support for esp32

forntoh, 2024-10-17:
Thanks for the feature request, I'm happy that you love the library even without trying it out, leaving a ⭐️ would be great 😊, I will investigate adding support for ARM boards until next weekend. My main focus was AVR boards, but the library is mature enough for it to expand to other boards.

closed 2024-10-20 by forntoh

forntoh, 2024-10-20:
@almofada1 FYI Support for ESP32 boards added in [v5.3.0](https://lcdmenu.forntoh.dev/reference/changelog.html#lcdmenu-v5-3-0)

almofada1, 2024-10-20:
Thank you very much. Life saver!