forntoh/LcdMenu #247: [BUG]: 4 bars appeared after update

almofada1, 2024-10-20:
### What happened?

After the update for support on the esp32, I've noticed that some bars appeared in the left side of the LCD
[image]

For context I'm using an Arduino Uno before the update and it was all fine. I updated the library and then they appeared, didn't change anything in the code.

### Version

5.3.0

### What environment are you seeing the problem in?

Arduino

### Board type

Arduino Uno

forntoh, 2024-10-20:
Hello, please can you attach your sketch file here (or link to repo/gist).

almofada1, 2024-10-20:
There. Just changed the pins and some things regarding Bluetooth

à(s) 21:22):

forntoh, 2024-10-20:
Unfortunately, I can't see it, please check what you attached. _(Remember that when replying by email attachments are not included, please post it directly on the issue)_

almofada1, 2024-10-20:
[lcdmenu.zip](https://github.com/user-attachments/files/17452179/lcdmenu.zip)

forntoh, 2024-10-21:
Seems not to be a bug, but an implementation issue, please refer to the RTOS example to see how the menu can be updated.

forntoh, 2024-10-21:
I haven't compiled this since I don't have your setup, however I have updated to code to closely match the desired workflow for using LcdMenu.

For now, I'll close the issue for now since it apparently happens only with your sketch. Please reopen it if you try agian and are convicted that it's a bug with the library.

```c++
#include <BluetoothSerial.h>
#include <DHT.h>
#include <ItemCommand.h>
#include <ItemSubMenu.h>
#include <LcdMenu.h>
#include <MenuScreen.h>
#include <SimpleRotary.h>
#include <display/LiquidCrystal_I2CAdapter.h>
#include <input/SimpleRotaryAdapter.h>
#include <renderer/CharacterDisplayRenderer.h>
#if defined(ARDUINO_ARCH_ESP32) || defined(ARDUINO_ARCH_ESP8266)
#else
#include "utils/printf.h"
#endif

BluetoothSerial SerialBT;

#define RX_PIN 16
#define TX_PIN 17
#define DHTPIN 18
#define DHTTYPE DHT11

DHT dht(DHTPIN, DHTTYPE);

#define MAX_DEVICES 5
String devices[MAX_DEVICES];
int deviceCount = 0;

#define LCD_ROWS 2
#define LCD_COLS 16

// Declare the callbacks
void connectBluetooth();
void displayTempHum();

// clang-format off
MENU_SCREEN(blueScreen, blueItems,
  ITEM_COMMAND("Connect", connectBluetooth));

MENU_SCREEN(tempScreen, tempItems,
  ITEM_BASIC("Temp:"),
  ITEM_BASIC("Hum:"));

MENU_SCREEN(mainScreen, mainItems,
  ITEM_SUBMENU("Bluetooth", blueScreen),
  ITEM_SUBMENU("Temp&Hum", tempScreen));
// clang-format on

float temp = 0;
float hum = 0;

LiquidCrystal_I2C lcd(0x27, LCD_COLS, LCD_ROWS);
LiquidCrystal_I2CAdapter lcdAdapter(&lcd);
CharacterDisplayRenderer renderer(&lcdAdapter, LCD_COLS, LCD_ROWS);
LcdMenu menu(renderer);
SimpleRotary encoder(27, 26, 25);
SimpleRotaryAdapter rotaryInput(&menu, &encoder);

void setup() {
    Serial.begin(115200);
    SerialBT.begin("ESP32Test");
    renderer.begin();
    dht.begin();

    // Create the degree symbol
    byte degree[8] = {B11100, B10100, B11100, B0000,
                      B00000, B00000, B00000, B00000};
    lcdAdapter.createChar(3, degree);

    menu.setScreen(mainScreen);
    Serial.print(SerialBT.available());
}

void loop() {
    // Observe rotary input for menu navigation
    rotaryInput.observe();

    // If there is data available from Serial2 (Bluetooth), write it to Serial
    // (USB)
    if (Serial2.available()) {
        Serial.write(Serial2.read());
    }

    // If there is data available from Serial (USB), write it to Serial2
    // (Bluetooth)
    if (Serial.available()) {
        Serial2.write(Serial.read());
    }

    // Check if there is any data available to read from the DHT sensor and
    // update the display
    displayTempHum();
}

void connectBluetooth() {
    Serial2.println("AT+INQ");  // Start inquiry
    Serial.print(Serial2.available());
}

void displayTempHum() {
    // Read the temperature from the DHT sensor
    float localTemp = dht.readTemperature();
    if (isnan(localTemp)) {
        // If the reading failed, display an error message
        tempItems[0]->setText("Error reading temperature");
    } else if (localTemp != temp) {
        // If the reading is successful and different from the last reading,
        // update the display
        char buf[10];
        snprintf(buf, 10, "Temp: %2.1f\003C", localTemp);
        tempItems[0]->setText(buf);
        temp = localTemp;
    }

    // Read the humidity from the DHT sensor
    float localHum = dht.readHumidity();
    if (isnan(localHum)) {
        // If the reading failed, display an error message
        tempItems[1]->setText("Error reading humidity");
    } else if (localHum != hum) {
        // If the reading is successful and different from the last reading,
        // update the display
        char buf[10];
        snprintf(buf, 10, "Hum: %2.1f%%", localHum, 3);
        tempItems[1]->setText(buf);
        hum = localHum;
    }
}
```

almofada1, 2024-10-21:
Thank you very much, I'll address the issue in a moment.

forntoh, 2024-10-27:
@almofada1 I was able to reproduce the bug, will fix it soon.

almofada1, 2024-10-27:
Oh nice, It doesn't really bother me but thank you! Happy to help.

à(s) 10:40):

closed 2024-10-27 by forntoh

forntoh, 2024-10-27:
@almofada1 Fixed in [v5.3.1](https://lcdmenu.forntoh.dev/reference/changelog.html#lcdmenu-v5-3-1), thanks for reporting, the version should be available on Arduino Library Manager in about an hour. It's already available on PlatformIO.

PR #257 (2024-10-27): Bugfix: 4 bars appear when no indicators are being shown
fixes #247

## Summary by CodeRabbit

- **New Features**
	- Enhanced workflow for publishing releases, improving versioning and release tagging.
- **Bug Fixes**
	- Adjusted character rendering logic for better visibility of up and down indicators.
- **Documentation**
	- Updated documentation for `CharacterDisplayRenderer` to clarify reserved slots for arrow icons and available custom character slots.
commit 28f57e5a (2024-10-27): Bugfix: 4 bars appear when no indicators are being shown (#257)

fixes #247

- **New Features**
- Enhanced workflow for publishing releases, improving versioning and
release tagging.
- **Bug Fixes**
- Adjusted character rendering logic for better visibility of up and
down indicators.
- **Documentation**
- Updated documentation for `CharacterDisplayRenderer` to clarify
reserved slots for arrow icons and available custom character slots.