forntoh/LcdMenu #273: Cannot Exit from Edit Mode

scorpion110374, 2024-12-07:
Hi I need your help
I am trying to use a Keypad 4x4 to mamage the menu.
But I cannot manage to exit edit mode with "#" and clears the value
This is my code:

#include <Keypad.h>
#include <ItemInput.h>
#include <LcdMenu.h>
#include <MenuScreen.h>
#include <display/LiquidCrystal_I2CAdapter.h>
#include <renderer/CharacterDisplayRenderer.h>

#define LCD_ROWS 2
#define LCD_COLS 16

//Define System Variables
int num = 0;
String empCode;
String numBulto;

//Declare Keypad variables
const byte ROWS = 4; //four rows
const byte COLS = 4; //four columns

char keys[ROWS][COLS] = {
  {'1','2','3','A'},
  {'4','5','6','B'},
  {'7','8','9','C'},
  {'*','0','#','D'}
};

// For Arduino Microcontroller
byte rowPins[ROWS] = {9, 8, 7, 6}; 
byte colPins[COLS] = {5, 4, 3, 2}; 

Keypad keypad = Keypad(makeKeymap(keys), rowPins, colPins, ROWS, COLS);

// Declare the call back function
void codigoInputCallback(char* value);
void bultoInputCallback(char* value);

// clang-format off
MENU_SCREEN(mainScreen, mainItems,
    ITEM_INPUT("Codigo", codigoInputCallback),
    ITEM_INPUT("Bulto", bultoInputCallback));
// clang-format on

LiquidCrystal_I2C lcd(0x3F, LCD_COLS, LCD_ROWS);
CharacterDisplayRenderer renderer(new LiquidCrystal_I2CAdapter(&lcd), LCD_COLS, LCD_ROWS);
LcdMenu menu(renderer);
//KeyboardAdapter keyboard(&menu, &Serial);

void setup() {
    Serial.begin(9600);
    renderer.begin();
    menu.setScreen(mainScreen);
}

void loop() {
    char key = keypad.getKey();
    switch (key){
      case NO_KEY:
        break;
      case '0': case '1': case '2': case '3': case '4':
      case '5': case '6': case '7': case '8': case '9':
        lcd.print(key);
        num = num * 10 + (key - '0');
        break;
      case 'A':
          menu.process(UP);
          break;
      case 'B':
          menu.process(DOWN);
          break;
      case 'C':
          menu.process(RIGHT);
          break;
      case 'D':
          menu.process(LEFT);
          break;
      case '*':
          menu.process(BACK);
          break;
      case '#':
          menu.process(ENTER);
          break;
    }
}

/**
 * Define callback
 */
void codigoInputCallback(char* value) {
    Serial.print("Codgigo CallBack");
    Serial.println(value);
}

void bultoInputCallback(char* value){
  Serial.print("Bulto CallBack"); 
};

scorpion110374, 2024-12-17:
Figured out how to exit "Edit Mode"
But, when change to next input, clear previuos input´s value
Need help please!

forntoh, 2024-12-18:
@scorpion110374 Is the original issue still valid? If yes, please re-open the issue. I'll close it for now.

closed 2024-12-18 by forntoh