forntoh/LcdMenu #280: [NEW-FEATURE] - Item Basic but can display variables

almofada1, 2024-12-22:
### Is your feature request related to a problem?

I'm struggling to find a way to display variables.

### Describe the solution you'd like?

An item just like Item basic or a improved version of item basic that accepts variables and can refresh its value.

forntoh, 2025-01-04:
Hi @almofada1 Can you please try out the PR and confirm if it matches your needs.

almofada1, 2025-01-04:
What do you mean by PR?

forntoh, 2025-01-04:
@almofada1  **Pull Request**, you can see it linked to this issue _(Just above my first comment and in the right side bar)_

The pull request contains the feature request you made, you can download the branch and try it out if it covers all the requirements you expect, if you have requests/questions you can also add comments on the PR.

forntoh, 2025-01-04:
@coderabbitai provide a detailed explanation to @almofada1 how to download the PR and test it. There's a starter example in `examples/ItemValue/ItemValue.ino`

closed 2025-01-04 by forntoh

almofada1, 2025-01-04:
Is there any way to update the values while I'm browsing the menu? I haven't tested it yet, just wondering.

PR #285 (2025-01-04): Add ItemValue template class and polling support for menu items
Closes #280

## Summary by CodeRabbit

## Release Notes

- **New Features**
  - Introduced a new `ItemValue` template class for dynamic menu item value display.
  - Added polling functionality to the menu system for real-time updates.
  - Integrated DHT22 sensor support for temperature readings.

- **Improvements**
  - Enhanced menu screen with interval-based refresh capabilities.
  - Added support for formatting and displaying dynamic values in menu items.
  - Updated documentation for the new value display item and polling mecha [... cut]
commit 6b1669d1 (2025-01-04): Add ItemValue template class and polling support for menu items (#285)

Closes #280

## Summary by CodeRabbit

## Release Notes

- **New Features**
- Introduced a new `ItemValue` template class for dynamic menu item
value display.
- Added polling functionality to the menu system for real-time updates.
  - Integrated DHT22 sensor support for temperature readings.

- **Improvements**
  - Enhanced menu screen with interval-based refresh capabilities.
- Added support for formatting and displaying dynamic values in menu
items.
- Updated documentation for the new value display item and polling
mecha [... cut]