forntoh/LcdMenu #287: [BUG]: The display timer is not updated by user input (button presses) that does not cause the menu to be redrawn.

ddevries6161, 2025-01-04:
### What happened?

Issue 1) The display timer is not reset by users input (button presses) that do not lead to the menu being redrawn. This can cause the display to turn off unexpectedly and not turn back on despite buttons being pressed.  For example, if a WIDGET_LIST has cycle set to false, if the UP button is pressed when at the top of the list or the DOWN button is pressed when at the bottom of the list, the display timer is not reset because MenuScreen::draw is never called.  In my opinion, the display timer should be reset whenever user input is received -- even if the user input doesn't result in the menu being redrawn.

Issue 2) The display timer will fail to turn off (hide) the display if updateTimer is called too infrequently (< 1kHz).  This is because in MenuRenderer.h, the updateTimer() function checks for an exact millisecond match of the current time to the timeout time.  If a call to updateTimer() doesn't occur in that exact millisecond, the display will remain on. 

I'm a novice with Arduino and github.  I don't know how to properly make a branch, make and document the changes, do a pull request, etc.  However, I'd be happy to send the modified source files to someone more familiar than me with github if that would be helpful.

I'm not sure if this is the best way to fix these bugs, but the following changes worked for me:
1) In MenuRenderer.h change:
    if (millis() != startTime + DISPLAY_TIMEOUT)...  to
    if (**millis() < startTime** + DISPLAY_TIMEOUT)...
Side effect: Causes display->hide to be called repeatedly after the timeout rather than just once, which is undesirable. 
To address this, in DisplayInterface.h, make the following bolded **additions**:
virtual void show() **{visible = true; }**
virtual void hide() **{visible = false; }**
**bool isVisible()   { return visible; }**
**private: 
bool visible = false;**

2) In the display adapters LiquidCrystalAdapter.h & SSD1803A_I2CAdapter.h, the show() and hide() functions need to check isVisible and act accordingly.  For example, in LiquidCrystalAdapter.h add the bolded if statement logic:
void show() override {
**if (!DisplayInterface::isVisible()) {**
lcd->display();
**DisplayInterface::show();}**}
...and...
void hide() override {
**if (DisplayInterface::isVisible()) {**
lcd->noDisplay();
**DisplayInterface::hide();}**} 

3) In MenusScreen.cpp delete or comment out **renderer->restartTimer();**

### Version

5.4.1

### What environment are you seeing the problem in?

Arduino

### Board type

ESP32-S3

ddevries6161, 2025-01-04:
I forgot one very important change to fix the bugs with the display timer.  In ButtonAdapter.h in the observe() function add the following inside the if (button->pressed()) logic...
menu->getRenderer()->restartTimer();

I'm not sure if the call to restartTimer() is also needed in the observe() functions in the other input adapters...I'm only using ButtonAdapter and haven't studied the others...

forntoh, 2025-01-05:
@ddevries6161 Thanks for catching this. There's an easier way to fix the timeout issue when no draw operation happens, that will not depend on the input method. I'll submit the PR soon.

forntoh, 2025-01-05:
Can you please open a separate issue for this 👇

> Issue 2) The display timer will fail to turn off (hide) the display if updateTimer is called too infrequently (< 1kHz). This is because in MenuRenderer.h, the updateTimer() function checks for an [...]

PR #290 (2025-01-05): Display timeout works only when a draw operation happens
Fixes #287

## Summary by CodeRabbit

- **Bug Fixes**
	- Adjusted timer restart mechanism in menu rendering to improve timing control and synchronization.
commit 486eca3f (2025-01-05): Bugfix: Display timeout works only when a draw operation happens

Fixes #287

- Added timer restart in the process method.
- Removed redundant timer restart from the draw method.
- Ensures display timeout functions correctly during user interactions.
commit b68035d2 (2025-01-05): Display timeout works only when a draw operation happens (#290)

Fixes #287

## Summary by CodeRabbit

- **Bug Fixes**
- Adjusted timer restart mechanism in menu rendering to improve timing
control and synchronization.
closed 2025-01-05 by forntoh