forntoh/LcdMenu #291: [NEW-FEATURE] - Add items during runtime

almofada1, 2025-01-05:
### Is your feature request related to a problem?

While developing one of my projects, I realised I would need to display the names of devices that would be discovered during runtime. So various submenus would be created as soon a device was detected and inside it there would be x items.

### Describe the solution you'd like?

The only solution I can think of is to use constructors to create either menus, submenus or items. By doing this, a dynamic menu could be created and there would be more room for development. Adding a function to remove dynamic items would be appreciated as well.

### Describe alternatives you've considered

Tried using constructors but no luck.

### Additional context

I might not know if this feature already exists but if it doesn't I'd really appreciate it as it would be very helpful to the project.

forntoh, 2025-01-05:
You can always create a screen dynamically like this

```c++
MenuItem* items[] = { 
    ITEM_BASIC("Test1"),
    ITEM_BASIC("Test2"),
    nullptr
};
MenuScreen* screen = new MenuScreen(items)
```

And display the screen with

```c++
menu.setScreen(mainScreen);
```

don't forget to cleanup memory if it's no longer needed

```c++
delete[] items;
delete screen;
```

---

In the future, I'll add a `setScreen` method to `ITEM_SUBMENU` so you can update the screen for a sub-menu on runtime.

almofada1, 2025-01-06:
When would that be?

forntoh, 2025-01-06:
I’m not sure at the moment, it really depends on when I have some free time. When it comes to developing this library, nothing is ever set in stone. I usually check out a few issues or discussions whenever I get the chance and work on possible solutions. I maintain it simply because I enjoy it.

almofada1, 2025-01-06:
Ok, thank you very much!

closed 2025-03-31 by forntoh