forntoh/LcdMenu #300: Support for pcf8574

Badn1nja, 2025-01-23:
### Is your feature request related to a problem?

I wish to use this libary with a pcf8574 controlled 16x2 screen

there are two libaries created but they do not have 

https://github.com/blackhack/LCD_I2C

https://github.com/mathertel/LiquidCrystal_PCF8574

### Describe the solution you'd like?

Would it be a case of making an additional wrapper under src/display for this?

### Additional context

Appreciate the work!

Badn1nja, 2025-01-23:
Update - seems to work - I had to remove lot's of lcd functions, is there a way to inject lcd.print statements before renderer takes over the device too?

forntoh, 2025-01-24:
@Badn1nja Refer to `SSD1803A_I2CAdapter.h` it's an adapter for a different type of LCD controller, you can create a similar adapter for your controller and submit a PR. Would you like to take on this task 😃?

forntoh, 2025-01-24:
> Update - seems to work - I had to remove lot's of lcd functions, is there a way to inject lcd.print statements before renderer takes over the device too?

Yes, you have an instance of the lcd, so you can just print stuff after hiding the menu (which can be done in `setup`) either by delaying `menu.setScreen(mainScreen);` to after you're done printing what you want or leaving it and calling `menu.hide()`

Badn1nja, 2025-01-24:
Appreciated! I will have a play now

Badn1nja, 2025-01-24:
> Appreciated! I will have a play now

Managed to implement a new Adapter and that's worked as the libarys match pretty closely, I will close this as I've raised a PR 👍🏼 

Thanks for the reply regardless!!

closed 2025-01-24 by Badn1nja