forntoh/LcdMenu #305: How to fetch Widget range value located in sub item

astrontech, 2025-02-05:
### Is your feature request related to a problem?

this is my 

i want to use vsp,span ..... values on void loop()

tried menu.getItemAt(), menu[], menu[][] 

also tried static_cast<ItemWidget*>(settings1ScreenItem[1])->getValue();

but failed to get values please help......

MENU_SCREEN(mainScreen, mainItems,
    ITEM_SUBMENU("Set Point", settings1Screen),
    ITEM_SUBMENU("Input Parameter", settings2Screen)
 
    );

// Create submenu and precise its parent
MENU_SCREEN(settings1Screen, settings1ScreenItems,
    ITEM_WIDGET(
        "Set Point",
        [](int vsp) { Serial.println(vsp); },
        WIDGET_RANGE(230, 1, 0, 300, "%d", 1)),
//        WIDGET_COMMAND(vspx = vsp),
    ITEM_WIDGET(
        " Span +/-",
        [](int span) { Serial.println(span);},
        WIDGET_RANGE(0, 1, 0, 5, "%d", 1)),
    ITEM_SUBMENU("Main", mainScreen));

MENU_SCREEN(settings2Screen, settings2ScreenItems,
    ITEM_WIDGET(
        " Vin Low",
        [](int vil) { Serial.println(vil); },
        WIDGET_RANGE(200, 1, 180, 230, "%d", 1)),
    ITEM_WIDGET(
        "Vin High",
        [](int vih) { Serial.println(vih); },
        WIDGET_RANGE(240, 1, 230, 300, "%d", 1)),
    ITEM_WIDGET(
        "Vin Trip",
        [](int vit) { Serial.println(vit); },
        WIDGET_RANGE(2, 1, 0, 15, "%d", 1)),
    ITEM_SUBMENU("Main", mainScreen));

### Describe the solution you'd like?

to get values updated in LcdMenu

### Describe alternatives you've considered

tried menu.getItemAt(), menu[], menu[][] 

also tried static_cast<ItemWidget*>(settings1ScreenItem[1])->getValue();

Badn1nja, 2025-02-05:
`MENU_SCREEN(mainScreen, mainItems,
  ITEM_WIDGET(
    "Temp",
    [](int settemp) { setpoint = settemp; },
    WIDGET_RANGE(80, 5, 60, 130, "%dC", 1)),
  ITEM_COMMAND("Start", start_heating));` can you update the value directly? like in the example?

astrontech, 2025-02-05:
> `MENU_SCREEN(mainScreen, mainItems, ITEM_WIDGET( "Temp", [](int settemp) { setpoint = settemp; }, WIDGET_RANGE(80, 5, 60, 130, "%dC", 1)), ITEM_COMMAND("Start", start_heating));` can you update the value directly? like in the example?

Thank you very much 

solved my problem

closed 2025-02-05 by astrontech