forntoh/LcdMenu #337: MENU_SCREEN does not work when all items are WIDGET_RANGE

xenington, 2025-04-06:
### What happened?

I created a MENU_SCREEN which is a submenu of another screen. The screen contains 2 x WIDGET_RANGE. The screen opens and displays correctly, but nothing is selectable and the focus arrow does not move from the 1st item.
Adding ITEM_BASIC("")); does not help (see my other issue report).
I have another MENU_SCREEN which contains both RANGE and LIST items and this works fine. Also, a screen that contains 2 x WIDGET_LIST works fine.   

### Version

- [x] I have checked that I am using the latest version of the library.

### What environment are you seeing the problem in?

Arduino

### Board type

Teensy 4.1

### Relevant log output

```shell

```

forntoh, 2025-04-07:
Can't reproduce this issue on ESP32, STM32 and UNO, those are the boards I have. The first two are ARM based boards like the Teensy. Please share the full example so I can try to reproduce exactly. Also ensure that you've [properly migrated](https://lcdmenu.forntoh.dev/reference/migration/index.html#migration-from-v5-9-0-to-v5-10-0) your code from v5.9.0

xenington, 2025-04-07:
Hi, thank you for your reply. 
Here is the code I am using. It compiles without error on Arduino IDE 2.3.5 & Teensyduino 1.59.0 and mostly works great! However, the surveyTypeScreen, surveyGPSSetupScreen and surveyScanSetupScreen only work when ITEM_BASIC("") is added (I commented it out in ScanSetup for testing). The optionScreen doesn't work at all, with either ITEM_BASIC or ITEM_COMMAND added. 

Thank you for your efforts in creating this library and following up.

```
#include <LcdMenu.h>
#include <MenuScreen.h>
#include <ItemSubMenu.h>
#include <ItemWidget.h>
#include <ItemCommand.h>
#include <display/LiquidCrystal_I2CAdapter.h>
#include <renderer/CharacterDisplayRenderer.h>
#include <widget/WidgetBool.h>
#include <widget/WidgetList.h>
#include <widget/WidgetRange.h>
#include <Button.h>
#include <input/ButtonAdapter.h>

#define LCD_ROWS 4
#define LCD_COLS 20
#define LCD_ADDRESS 0x27

extern MenuScreen* calibrationScreen;
extern MenuScreen* surveyTypeScreen;
extern MenuScreen* surveyGridSetupScreen;
extern MenuScreen* surveyGPSSetupScreen;
extern MenuScreen* surveyScanSetupScreen;
extern MenuScreen* optionScreen;

void download();
void shutDown();
void calibrateMaxMin();
void calibrateOrientate();
void calibrateRotateBottom();
void calibrateAdjustBottom();
void startGridSurvey();
void startGPSSurvey();
void startScanSurvey();
void buzzerVolume();
void screenContrast();

MENU_SCREEN(mainScreen, mainItems,
    ITEM_SUBMENU("Calibration", calibrationScreen),
    ITEM_SUBMENU("Survey Type", surveyTypeScreen),
    ITEM_SUBMENU("Options", optionScreen),
    ITEM_COMMAND("Download", download),
    ITEM_COMMAND("Shut Down", shutDown));

MENU_SCREEN(calibrationScreen, calibrationItems,
    ITEM_COMMAND("Max/Min", calibrateMaxMin),
    ITEM_COMMAND("Orientation", calibrateOrientate),
    ITEM_COMMAND("Rotate Bottom", calibrateRotateBottom),
    ITEM_COMMAND("Adjust Bottom", calibrateAdjustBottom));

MENU_SCREEN(surveyTypeScreen, surveyTypeItems,
    ITEM_SUBMENU("Grid Survey", surveyGridSetupScreen),
    ITEM_SUBMENU("GPS Survey", surveyGPSSetupScreen),
    ITEM_SUBMENU("Scan Survey", surveyScanSetupScreen),
    ITEM_BASIC(""));

std::vector<const char*> listLinearReturn = {"Linear", "Return"};
std::vector<const char*> listStartCorner = {"NW", "NE", "SW", "SE"};
std::vector<const char*> listStartDirection = {"CW", "CCW"};
std::vector<uint8_t> listSensorNumber = {2, 4, 8, 16};
std::vector<uint8_t> listReadRate = {1, 2, 4, 8, 16, 32};
std::vector<uint8_t> listTraversePerMetre = {1, 2, 4};

uint8_t sensorNumber = 8;       // default 8 sensors (2, 4, 8 or 16)
uint8_t readRate = 32;          // default 32 per metre/second (1, 2, 4, 8, 16 or 32)
[... 110 lines cut ...]
void calibrateAdjustBottom()
{
  Serial.println("Adjust Bottom");
}
void startGridSurvey()
{
  Serial.println("start Grid");
}
void startGPSSurvey()
{
  Serial.println("start GPS");
}
void startScanSurvey()
{
  Serial.println("start Scan");
}
void buzzerVolume()
{
  Serial.println("buzzer volume");
}
void screenContrast()
{
  Serial.println("screen contrast");
}
```

forntoh, 2025-04-07:
Thanks for the code, I'll try compile it.

xenington, 2025-04-07:
Thank you for trying. 

Your video shows a 16x2 lcd. I am using a 20x4. Could that be the problem? I don't have a 16x2 to try unfortunately. The only other possibility is the Teensy 4.1. 

Anyway, I will have see if I can find a solution.

xenington, 2025-04-08:
I have discovered that no menu screen works unless all 4 rows on my 20x4 screen contain a menu item. It does not matter what the menu items are or whether the screens are main screen or sub screens. It was simply a coincidence that adding ITEM_BASIC("") made a total of 4 menu items on a particular screen. Less than 4 menu items and the screen does nothing.

On your 16x2, can you try just 1 menu item, i.e. 1 row? Thanks.

forntoh, 2025-04-08:
@xenington oh nice, I remember seeing this part of the code last week and thinking "I need to fix it", it's a bug and forgot to plan it. Thanks for reporting this, will get a fix sometime this week.

closed 2025-04-09 by forntoh