forntoh/LcdMenu #349: ITEM_VALUE have lost float compatibility

TBAMax, 2025-05-06:
### What happened?

ITEM_VALUE do not display **float** values. The text part of the formatting string is displayed, but nothing and zero space for value.

Unsigned long is displayed fine, so it seems to be specific issue with float type.

This this issue came in at least release 5.11.0, hard to say exactly at what point. I think 5.8.0 was fine previously.

`float U1 = 0.0;`
...
`ITEM_VALUE("V1", U1, "%9.3fV"), `
displays only: V1:V

Same for WIDGET_RANGE_REF
`WIDGET_RANGE_REF(temperature2, 0.1f, -30.0f, 100.0f, "%4.1fC", 0, false),`
displays only "C"

### Version

- [x] I have checked that I am using the latest version of the library.

### What environment are you seeing the problem in?

Platform IO

### Board type

STM32

### Relevant log output

```shell

```

ShishkinDmitriy, 2025-05-08:
Hi, 

As I understand it's not a bug, it's just expensive to have support of floats, so it was hidden.
See [utils/printf.h](https://github.com/forntoh/LcdMenu/blob/master/src/utils/printf.c#L525). Belive you just need to add some props: `USE_CUSTOM_PRINTF` and `PRINTF_SUPPORT_FLOAT`

Dmitry

forntoh, 2025-05-15:
@TBAMax can you please confirm if this brings back float printing, if so then this bug and PR can be closed.

You can create a new PR to add this new info in the docs / FAQs.

TBAMax, 2025-05-15:
> @TBAMax can you please confirm if this brings back float printing, if so then this bug and PR can be closed.
> 
> You can create a new PR to add this new info in the docs / FAQs.

I can confirm that the PR #350 that I made brings back the float capability on bluebill   STM32 board:bluepill_f103c8_128k
This one I tested on physical board.
Other STM32 boards (F7 series and such) I have not tested, so no idea if those work with or without the PR or both ways.

The compile time flags suggested by @ShishkinDmitriy I have not tried. I would prefer the built in method that may PR brings instead of the flags method, because with the flag method every new user with STM32 board needs to manually add flags to the patformio.ini and that seems unnessesary hassle.

forntoh, 2025-05-23:
@TBAMax concerning this,

> The compile time flags suggested by @ShishkinDmitriy I have not tried. I would prefer the built in method that may PR brings instead of the flags method, because with the flag method every new user with STM32 board needs to manually add flags to the patformio.ini and that seems unnessesary hassle.

I think this should be an opt-in feature as @ShishkinDmitriy suggested, not every user with a STM32 board would need float support and removing the header check would mean adding overhead in their existing code without float need (making the sketch size larger for them). If anyone desires float support, then they should use the platform available method (which is currently using the flag).

TBAMax, 2025-05-23:
I have tested another possibility to resolve the issue: To enable stm32 built in float support with the compiler flag
`-D PIO_FRAMEWORK_ARDUINO_NANOLIB_FLOAT_PRINTF`
This also works, but uses slightly more RAM (for my specific about 500B more), and slightly more flash (about 9kB more for my application). So the one included with the LcdMenu in the utilites folder seems to be better.

Also, to note, when both float supports (stm32 built in, and the one supplied with LcdMenu) are active simultaneously, the flash usage is even more higher.

So currently the custom `printf.c` and `printf.h` supplied with the LcdMenu are prefferred.

> You can create a new PR to add this new info in the docs / FAQs.

@forntoh Sorry, I did not understand at first that You prefer to simply add the
`build_flags = -DUSE_CUSTOM_PRINTF` to the documentation, and not restore the default behaviour at all.
Also in this case some examples that are using floats are compiled without actual float support! 
So actually, at the moment,I do not agree with this solution, so I do not rush making a new PR with it.

There is a possibility instead to use `-DPRINTF_DISABLE_SUPPORT_FLOAT` to save on memory for small micros. And have the default to float enabled...
I will let it sit a little, and think about it.

PR #350 (2025-05-27): Float printing for STM32
## Description
Remove STM32 from the list of known to have full sprintf support, because apparently testing shows on bluePill there is no float support by default.
Re allows custom printf for STM32, that was turned off recently most likely by accident. Surprise is that STM32 bluepill_f103c8_128k do not have float printing support by default.

Fixes #349

### Checklist

#### General Requirements

- [x] I have kept this PR in draft until all the required tasks are completed.
- [x] I have reviewed the [contributing guidelines](https://github.com/forntoh/LcdMenu/blob/maste [... cut]
commit 0909ebe3 (2025-05-27): Float printing for STM32 (#350)

## Description
Remove STM32 from the list of known to have full sprintf support,
because apparently testing shows on bluePill there is no float support
by default.
Re allows custom printf for STM32, that was turned off recently most
likely by accident. Surprise is that STM32 bluepill_f103c8_128k do not
have float printing support by default.

Fixes #349

### Checklist

#### General Requirements

- [x] I have kept this PR in draft until all the required tasks are
completed.
- [x] I have reviewed the [contributing
guidelines](https://github.com/forntoh/LcdMenu/bl [... cut]
closed 2025-05-27 by forntoh