forntoh/LcdMenu #376: BACKSPACE not working

almofada1, 2025-06-14:
### What happened?

I've tried to double press quickly and the characters won't get deleted.

### Version

- [x] I have checked that I am using the latest version of the library.

### What environment are you seeing the problem in?

Arduino

### Board type

ESP32

### Relevant log output

```shell

```

zkarolyi, 2025-06-14:
Using SimpleRotary?
it has:
`    unsigned int _debounceSDelay = 200;`

in LcdMenu:
`#define DOUBLE_PRESS_THRESHOLD 300`

Not easy to click more than 200 and less than 300 ms.
I have changed it in my code:
```
#ifdef DOUBLE_PRESS_THRESHOLD
#undef DOUBLE_PRESS_THRESHOLD
#endif
#define DOUBLE_PRESS_THRESHOLD 500
```

almofada1, 2025-06-19:
Doesn't solve the issue.

almofada1, 2025-06-20:
@forntoh how can I delete the last character in input charset item with SimpleRotary and SimpleRotaryAdapter?

almofada1, 2025-06-25:
@forntoh it's urgent can you please take a look?

forntoh, 2026-09-28:
Thanks for your patience on this. A fix is ready in #405: `SimpleRotaryAdapter` now enforces an effective double-press threshold of at least 500 ms — with SimpleRotary's ~200 ms button sampling, shorter windows could not reliably produce a BACKSPACE — while `DOUBLE_PRESS_THRESHOLD = 0` still disables double-press entirely. The practical guidance is also documented in the rotary examples.

The PR is awaiting merge. If you want to test it early, point PlatformIO at the `fix/376-simple-rotary-backspace` branch and let us know if double-press backspace works reliably with your setup.

closed 2026-09-28 by forntoh